robert-hh/SH1106 #1: Only 8 pixels high available

alex7088, 2017-04-14:
With Framebuffer1 I can use only 8 pixels high dipslay. When I changes from Framebuffer1 to Framebuffer problem solved

robert-hh, 2017-04-14:
Hello Alex. Thanks for the note. It is not clear to me what you mean with: 
> 8 pixels high dipslay
Do you mean monochrome displays or really a display with the geometry width x 8? I used the driver with a monochrome 128x64 pixel display. What kind of display are you using?
Besides that, I think the change you used is safe.

alex7088, 2017-04-15:
No this is common 128x64 but with original driver it shown only 128x8
Other part of display was off
I l send a photo

robert-hh, 2017-04-15:
Found the reason: The underlying class was changed, and the format parameter obsolete for Framebuffer1 was taken stride value. Anyhow, I updated the repository.
Thanks a lot.

closed 2017-04-15 by robert-hh