robert-hh/SH1106 #2: hardware scroll

pacmac, 2017-06-09:
Hi Robert;

Thanks for this driver, works great, is there a hardware scoll built in which would enable non-blocking vertical and horizontal scroll.

I tried adding this but it did not work:

    SET_HWSCROLL_OFF    = const(0x2e)
    SET_HWSCROLL_ON     = const(0x2f)
    SET_HWSCROLL_RIGHT  = const(0x26)
    SET_HWSCROLL_LEFT   = const(0x27)

    def right(self,start=0,stop=63):
      self.write_cmd(SET_HWSCROLL_RIGHT)
      self.write_cmd(0X00)
      self.write_cmd(start)
      self.write_cmd(0X00)
      self.write_cmd(stop)
      self.write_cmd(0X00)
      self.write_cmd(0XFF)
      self.write_cmd(SET_HWSCROLL_ON);

robert-hh, 2017-06-09:
The driver is using the framebuffer module, which just keeps one page. The built-in scrolling as well as the scrolling in framebuffer does a cyclic scroll, AFAIK. In order to scroll continouly with new content, you have to:
1. scroll the device
2. erase the part that is to be newly written
3. write that area
4. update the screen 
with hardware scrolling you could at least reduce the amount of data to be transferred to just the new data.
And yes, to do that in the background parallel to other operations requires some kind of semi-parallel execution.

pacmac, 2017-06-09:
This does a beautifully smooth scroll by using SET_START_LINE instead of SET_HWSCROLL_ON which I can't get to work at all:

    oled.write_cmd(SET_START_LINE | 0)
    oled.clear()
    
    oled.text('1234567890ABDCDEF', 0, 1, 1)
    oled.text('1234567890ABDCDEF', 0, 11, 1)
    oled.show()
    
    for x in range(63):
      oled.write_cmd(SET_START_LINE | x)
      time.sleep_ms(10)

robert-hh, 2017-06-09:
That's good. I use the same mechanism in the SSD1963 driver for PyBoard. But it will always show the full screen content, just rotated. When using with over functions, you have to respect the setting for START_LINE.  Not difficult, just a little bit of thinking forth and back. If you want to roll off content from the screen, you have to erase that. One addition I had in mind for the driver was options for a partial show().

diginfo, 2017-06-09:
This works very well to vertically scroll a whole page:

    def vpage(self,delay=10):
      show = False
      self.write_cmd(SET_START_LINE | 0)
      for x in range(self.height-1):
        self.write_cmd(SET_START_LINE | x)
        if not show:
          self.show()
          show = True
        time.sleep_ms(delay)
      self.fill(0)

I have only ever used the 0.96" displays but today received a batch of the 1.3" with the SH1106 vs SSD1306,.

I have always found it virtually impossible to read the text on the 0.96", but these 1.3" are definitely usable.

Thanks again for the driver.

closed 2017-06-14 by pacmac