robert-hh/SH1106 #8: AttributeError: 'str' object has no attribute 'init'

bgolab, 2019-12-03:
MicroPython v1.11 on 2019-05-29; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>>
>>> from machine import I2C, Pin
>>> i2c=I2C(1)
>>> i2c.scan()
[76]
>>> import sh1106
>>>
>>>
>>> display = sh1106.SH1106_I2C(128,32, i2c, 'none', 76)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sh1106.py", line 176, in __init__
AttributeError: 'str' object has no attribute 'init'

robert-hh, 2019-12-03:
Obviously not. The parameter has to be just `None`, not a string.

bgolab, 2019-12-03:
ok.
ssd1306 works fine
>>> from machine import I2C, Pin
>>> i2c=I2C(1)
>>> i2c.scan()
[60, 76]
>>> import ssd1306
>>> oled=ssd1306.SSD1306_I2C(128,32,i2c,60)
>>> oled.text('ala', 1, 1, 1)
>>> oled.show()

but (after re-powering the board to clear the I2C state):
>>> from machine import I2C, Pin
>>> i2c=I2C(1)
>>> i2c.scan()
[60, 76]
import sh1106
>>> display = sh1106.SH1106_I2C(128,32, i2c, None,60)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sh1106.py", line 177, in __init__
  File "sh1106.py", line 111, in __init__
  File "sh1106.py", line 117, in init_display
  File "sh1106.py", line 151, in show
  File "sh1106.py", line 190, in sw_write_data
OSError: I2C operation not supported

I need to dig it deeper...

robert-hh, 2019-12-03:
Are you using a SH1106 or an SSD1306?

bgolab, 2019-12-03:
I use SH1106 (at least the vendor claims that). That's why I need to investigate it deeper.

I started with your driver, then switched to SSD1306 which works fine.

closed 2019-12-04 by robert-hh