robert-hh/SH1106 #10: Using default I2C pins

robhamerling, 2020-08-16:
I'm trying to use a SH1106 display on a "black pill" (STM32F411CEU6). The display is connected to the first I2C module. When I specify i2c=I2C(1) - which is a supported method according to the machine.I2C docs - I get a write_data error:  

  "sh1106.py", line 190, in sw_write_data 
  OSError: I2C operation not supported

I do not understand why I get this error, since it is a supported method of machine.I2C.
With a print(i2c) I see:

  I2C(1, scl=B6, sda=B7, freq=480000)

Which shows that the proper pins are used (the default pins).
When I specify these pins explicitly:
 
 i2c=I2C(scl=Pin("PB6", sda=Pin("PB7"))

the program/library doesn't complain. Why cannot I simply use I2C(1) ?

BTW: The reason that I do not like to specify the pins is that I want to explore 'alternate pin functions' moving the default I2C pins to e.g. PB8/PB9. Note: when I specify B8/B9 (without using the alternate pin functions) then the display works also fine. Looks like software-I2C is used.

robert-hh, 2020-08-16:
Yes, the difference between SW and HW I2C may cause that. The HW-I2C does not support all methods. I have to look into the driver. It is rather old, and the firmware has evolved since then. I should drop the differentiation between SW and HW mode.

robert-hh, 2020-08-16:
Since you have a set-up running, you could try using the simplified I2C class below:
```
class SH1106_I2C(SH1106):
    def __init__(self, width, height, i2c, res=None, addr=0x3c,
                 external_vcc=False):
        self.i2c = i2c
        self.addr = addr
        self.res = res
        self.temp = bytearray(2)
        if res is not None:
            res.init(res.OUT, value=1)
        super().__init__(width, height, external_vcc)

    def write_cmd(self, cmd):
        self.temp[0] = 0x80  # Co=1, D/C#=0
        self.temp[1] = cmd
        self.i2c.writeto(self.addr, self.temp)

    def write_data(self, buf):
        self.i2c.writeto(self.addr, b'\x40'+buf)
```

robhamerling, 2020-08-16:
Yes, this modification works fine with: i2c= I2C(1)!
I tested successfully not only your sample program, but also a more extensive test-pgm of my own, with several elementary text and graphic functions.  
I was also curious if the 'alternate function' facility for Pins would now work and I'm happy to inform you that using alternate default I2C pins B8/B9  (in stead of B6/B7) works fine,  while still specifying simply i2c=I2C(1).

[maybe for another issue?]
The SH1106 datasheet tells me that this chip supports 132x64 pixels, while your comments in the library and your sample seem to suggest that only 128x64 is supported. Both 128 and 132 work fine.

robert-hh, 2020-08-16:
Thank you for the feedback. Then I will change the repository.

closed 2021-11-07 by robert-hh