robert-hh/SH1106 #11: Use this project for Python-Raspberry Pi

tik-tak-soft, 2021-01-21:
Hi, I was really happy to see this directory, very much I am looking for a screen with driver sh1122 lib.
But I was sorry to see that it was not for Python but for Micro Python. Do you think you have the option to match Python?
Or I would fit in if it weren't too complicated.

robert-hh, 2021-01-21:
This is the driver for SH1106. And it is based on on the framebuffer class of MicroPython. The effort for porting it should be comparable to writing a new driver.

tik-tak-soft, 2021-10-09:
> @tik-tak-soft <https://github.com/tik-tak-soft> check out this
> <http://www.waveshare.com/w/upload/0/0f/1.3inch-OLED-Code.7z> code
> package from waveshare, an oled manufacturer, for the sh1106. it has
> examples for the raspi too

Thank you! I greatly cherish your effort to worry for me to solve! I hope
that it will give me the necessary response!
👍
[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality10&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality10&>
09.10.21,
21:28:18

‫בתאריך יום ו׳, 8 באוק׳ 2021 ב-16:58 מאת ‪KohliNaman‬‏ <‪
***@***.***‬‏>:‬

closed 2021-11-07 by robert-hh