robert-hh/SH1106 #12: sh1107 compatibility

michalsprawka, 2021-02-15:
Will library works with sh1107 controller?

robert-hh, 2021-02-15:
I do not know. I do not have such a device.

closed 2021-02-16 by robert-hh

PaulskPt, 2021-12-23:
With small adjustments yes. I use a RPi Pico with a SH1107 with SPI interface. See issue #17 for the description of my hardware, the SPI call and an image (picture).
I put the SH1106.py script in a separate file and called it from another test script: 
My test script is as follows: 
```
from machine import Pin, SPI
from sh1106 import SH1106, SH1106_SPI

# Pins for a Raspberry Pi and SH1107 on a Pimoroni Pico Breakout Garden base:
dc = 16 #board.GP16
cs = 17 #board.GP17  # for BG SPI socket A (front). GP22 for socket B
sck = 18 #board.GP18
mosi = 19 #board.GP19
rst = 20 #board.GP20  # for BG SPI socket A (front). GP21 for socket B

spi = SPI(0, 100000, mosi=Pin(mosi), sck=Pin(sck))
display = SH1106_SPI(128, 128, spi, Pin(dc), Pin(rst), Pin(cs), 0)
display.sleep(False)
display.fill(0)
display.text('Testing 1', 0, 0, 1)
display.show()
```