robert-hh/SH1106 #13: Pi Pico

smiffycom, 2021-04-06:
Hi, i can't get this driver to work on the Pi Pico

I use;  display = sh1106.SH1106_I2C(128, 64, i2c, Pin(16), 0x3c)

all I get is;

File "sh1106.py", line 21, in <module>
AttributeError: 'module' object has no attribute 'SH1106_I2C'

any ideas?

robert-hh, 2021-04-06:
No. If you use e.g.:
```
import sh1106
from machine import I2C
i2c=I2C(0)
display = sh1106.SH1106_I2C(128, 64, i2c, Pin(16), 0x3c)
```
That should work. What is the result of:
```
import sh1106
dir(sh1106)
```

closed 2021-04-07 by robert-hh

smiffycom, 2021-04-10:
Hi Robert,

I have tried LOTS of times to use your library with no success. I have placed your code in /lib and the root of my Pico. neither works. I have tried nearly every combination imaginable it always comes back with

Traceback (most recent call last):
  File "<stdin>", line 13, in <module>
  File "sh1106.py", line 39, in <module>
ImportError: can't import name SH1106_I2C

I'm using  this I2C declaration
i2c = I2C(1,scl=Pin(19), sda=Pin(18), freq=400000)
As this is where the device is connected, but I have tried the default.. still no joy

here's the results of a DIR

>>> import sh1106
>>> dir (sh1106)
['__class__', '__name__', 'I2C', 'Pin', 'sh1106']

where to now?

Regards,

smiffy

Closed #13<https://github.com/robert-hh/SH1106/issues/13>.

robert-hh, 2021-04-10:
a) did you download the sh1106.py from this place: https://github.com/robert-hh/SH1106
b) what else is in the file system of your board?
c) What is the name of the file you set up to test the driver. Is it by chance sh1106.py too? That must not be. You have to give it a different name.

smiffycom, 2021-04-10:
a) Yes, downloaded it a dozen times
b) just drivers for a few totally unrelated bits of hardware, and a few bits of code - there's loads of RAM available
c) yes, your code is in /lib with edits for my I2C hardware, the program in /root is another name

in your code as posted;
i2c = I2C(scl=Pin(5), sda=Pin(4), freq=400000)
I get
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
  File "sh1106.py", line 5, in <module>
TypeError: 'id' argument required
Line 5 being;
i2c = I2C(scl=Pin(5), sda=Pin(4), freq=400000)
if I add the (0,
i2c = I2C(0,scl=Pin(5), sda=Pin(4), freq=400000)
I get the familiar

Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
  File "sh1106.py", line 6, in <module>
AttributeError: 'module' object has no attribute 'SH1106_I2C'
Line 6 being;
display = sh1106.SH1106_I2C(128, 64, i2c, None, 0x3c)

smiffy

robert-hh, 2021-04-10:
My code contains just an example of how to use the driver as comments. Obviously you have to adapt that for the target hardware. The initial example was for ESP8266. Are you sure that there is no other file called sh1106.py or sh1106.mpy in the file system? What is the value of sys.path? And, which firmware do you use?  And to be sure, the driver must look like the lines below:
```
#
# MicroPython SH1106 OLED driver, I2C and SPI interfaces
#
# The MIT License (MIT)
#
# Copyright (c) 2016 Radomir Dopieralski (@deshipu),
#               2017 Robert Hammelrath (@robert-hh)
#
# Permission is hereby granted, free of charge, to any person obtaining a copy
# of this software and associated documentation files (the "Software"), to deal
# in the Software without restriction, including without limitation the rights
# to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
# copies of the Software, and to permit persons to whom the Software is
# furnished to do so, subject to the following conditions:
#
# The above copyright notice and this permission notice shall be included in
# all copies or substantial portions of the Software.
#
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
# AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
# OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
# THE SOFTWARE.
#
# Sample code sections
# ------------ SPI ------------------
# Pin Map SPI
#   - 3v - xxxxxx   - Vcc
#   - G  - xxxxxx   - Gnd
#   - D7 - GPIO 13  - Din / MOSI fixed
#   - D5 - GPIO 14  - Clk / Sck fixed
#   - D8 - GPIO 4   - CS (optional, if the only connected device)
#   - D2 - GPIO 5   - D/C
#   - D1 - GPIO 2   - Res
#
# for CS, D/C and Res other ports may be chosen.
#
# from machine import Pin, SPI
# import sh1106

# spi = SPI(1, baudrate=1000000)
# display = sh1106.SH1106_SPI(128, 64, spi, Pin(5), Pin(2), Pin(4))
# display.sleep(False)
# display.fill(0)
# display.text('Testing 1', 0, 0, 1)
# display.show()
#
# --------------- I2C ------------------
#
# Pin Map I2C
#   - 3v - xxxxxx   - Vcc
#   - G  - xxxxxx   - Gnd
#   - D2 - GPIO 5   - SCK / SCL
#   - D1 - GPIO 4   - DIN / SDA
#   - D0 - GPIO 16  - Res
#   - G  - xxxxxx     CS
#   - G  - xxxxxx     D/C
#
# Pin's for I2C can be set almost arbitrary
#
# from machine import Pin, I2C
# import sh1106
#
# i2c = I2C(scl=Pin(5), sda=Pin(4), freq=400000)
# display = sh1106.SH1106_I2C(128, 64, i2c, Pin(16), 0x3c)
# display.sleep(False)
[... 131 lines cut ...]
        if self.cs is not None:
            self.cs(1)
            self.dc(0)
            self.cs(0)
            self.spi.write(bytearray([cmd]))
            self.cs(1)
        else:
            self.dc(0)
            self.spi.write(bytearray([cmd]))

    def write_data(self, buf):
        self.spi.init(baudrate=self.rate, polarity=0, phase=0)
        if self.cs is not None:
            self.cs(1)
            self.dc(1)
            self.cs(0)
            self.spi.write(buf)
            self.cs(1)
        else:
            self.dc(1)
            self.spi.write(buf)

    def reset(self):
        super().reset(self.res)
```