robert-hh/SH1106 #17: Two omissions and one error in SPI example

PaulskPt, 2021-12-23:
Hi Robert,
Thank you for your repo. 
I am using a RPi Pico on a Pimoroni Pico Breakout Garden base ```PIM549```and a Pimoroni OLED SPI 1.12in 128x128px monochrome (SH1107) ```PIM473```.
For this SH1107 there are several Circuitpython solutions and solutions to run on a Raspberry Pi (Raspberry Pi Operating System). But all these use various libraries with a lot of dependencies. So I was looking for a Micropython solution. I google'd and found your repo.
In your README.md, in the remarks about SPI you wrote: ```SCLK, MOSI, D/C are always required. ```.
In the example for SPI in your sh1106.py script, line 44, you wrote: ```spi = SPI(1, baudrate=1000000)```. 
TMHO there are in this call to SPI() : a) two omissions and b) one error.
ad a) missing parameters for mosi and sclk;
ad b) baudrate must be 100 kHz instead of your 1 MHz.
For my hardware the call to SPI, first parameter has to be a value 0. So, for my hardware the call to SPI is: ```spi = SPI(0, 100000, mosi=Pin(mosi), sck=Pin(sclk))```.
And with this it works for my SH1107. See the image:

[image]

robert-hh, 2021-12-23:
a) `SCLK, MOSI, D/C` mention the signals of the Display, not the signal of the SPI bus.
b) Some boards (like the ESP8266, PyBoard or the mimxrt board) have fixed pins for SPI resp. hardware SPI modules. Then the pins cannot be set. The SPI module id may also differ between boards. Some start at 0 (like RP2), some at 1. The script was initially made for a project with an ESP8266. I could indeed upgrade the examples.
In any case, it is good to know that it works with the sh1107 display.

PaulskPt, 2021-12-23:
OK, thank you for the explanation!

closed 2021-12-23 by PaulskPt