robert-hh/SH1106 #18: i2c not working on certain brand of 1.3" oled display

erjdriver, 2022-02-02:
Hi,

Using this with the PICO and the library works fine with one brand of 1.3" display.

But it doesn't work with another brand.

https://www.ebay.com/itm/163020994321

I believe it's sh1106 - same code..same wiring...I just swap the units and restart.

The test code I used was your sample at the top of the lib source code.

robert-hh, 2022-02-02:
EIO usually means that the I2C connection fails. Look at the soldering on the back side. The default address used by the driver is 0x3c, which matches the 0x78 on the board. If the jumper is at 0x7a, use the address 0x3d. You can as well check the I2C bus for active devices using:

    from machine import I2C
    i2c=I2C(..... whatever your settings are ..)
    i2c.scan()

If there are active I2C device detected, the scan returns a list of addresses as decimal values. 0x3c = 60, 0x3d = 61.

erjdriver, 2022-02-02:
Thanks for the very quick response.

I did do a scan just after the post and it said nothing found - so most probably not a lib issue.

Thanks - for your contribution.

robert-hh, 2022-02-02:
No. If the device does not respond to i2c.scan(), then it is a communication issue. I cannot tell from the pictures whether it has pull-up resistors. The picture shows quiet a freq 4k7Ohm resistors, but I do not know if the picture matches your device. You could try to add some or run the scan test at a low I2C frequency, like 1 kHz, just to be sure.
If it has a Voltage regulator, you can try to supply it with 5V. The logic levels are still at 3.3V.

erjdriver, 2022-02-02:
I've had some luck - reducing the frequency and scanning - the device is found.

I also put it back to 400,000 and it works also.

but it's still not displaying.

I've got the following:

display = SH1106_I2C(128, 64, i2c, None, addr=0x3c)

I'm not setting the reset pin - what should that pin connect to?

robert-hh, 2022-02-02:
The display does not seem to have a reset pin. If it is a sh1106, it has no reset command. But anyhow, communication has to be stable. The fact that scan() works, does not guarantee that other communication is fine. For scan, just a single bye is sent, the address. Longer messages may still fail.
Does the rear side of the module match the picture on EBay?

erjdriver, 2022-02-02:
yes it does - the address is 0x3c. no extra reset pin.

`from machine import Pin, I2C
from sh1106 import SH1106_I2C

i2c = I2C(0,scl=Pin(9), sda=Pin(8), freq=50000)

print('Scan i2c bus...')
devices = i2c.scan()

if len(devices) == 0:
  print("No i2c device !")
else:
  print('i2c devices found:',len(devices))

  for device in devices:  
    print("Decimal address: ",device," | Hexa address: ",hex(device))

#        self.display = SH1106_I2C(128, 64, i2c, None, addr=0x3c)

display = SH1106_I2C(128, 64, i2c, None, addr=0x3c)
display.sleep(False)
display.fill(0)
display.text('Testing 1', 0, 0, 1)
display.show()
print("OK")
`

It finds the device every once in a while...

fails in the constructor often on writecmd() and every once in a while the whole code finishes but no display.

i swap out the units and restart and the other module works fine.

note i've reduced the freq down to 50000.

closed 2022-02-02 by erjdriver