robert-hh/SH1106 #20: Pi Pico UV sensor(veml6070)

Kratos800, 2022-03-18:
The code below is my code on pi pico
I have installed all the libraries needed.

import digitalio
import time
import board
import radio
import bmp280
import UV

led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT
led.value = True
cansat_temperature = bmp280.read_temperature()
cansat_pressure = bmp280.read_pressure()
room_temp = cansat_temperature
room_pressure = cansat_pressure

print("T: {:.3f} P: {:.2f}".format(cansat_temperature, cansat_pressure))

while True:
    led.value = not led.value
    print("Radio message sent")
    time.sleep(1)
    cansat_temperature = bmp280.read_temperature()
    print(cansat_temperature)
    cansat_pressure = bmp280.read_pressure()
    print(cansat_pressure)

    radio.send(
        "[CGS] Temperature: {:.2f} Pressure {:.0f}".format(room_temp, room_pressure)
    )

I am trying to code veml6070 UV sensor and the code is:
import board
import adafruit_veml6070

i2c = board.i2c()
but I am getting error:
AttributeError: 'module' object has no attribute 'i2c'

robert-hh, 2022-03-18:
What is the relation of you post to the SH1106 driver?

And not related to that: it seems as if you use CircuitPython. I cannot tell you more about CircuitPyhton than you can find in it's documentation. The error you get is a CircuitPython question. You can get the list of object in the board module with:

dir(board)

It shows that the name of the object is I2C, not i2c. So the line should most probably be:

i2c=board.I2C()

Using the default SDA and SCL pins.

Kratos800, 2022-03-19:
I already tried to do that but still it was showing the same error

robert-hh, 2022-03-19:
What is the output of:

import board
dir(board)

Kratos800, 2022-03-19:
code.py output:

Code done running.

robert-hh, 2022-03-19:
That#s not the output of the dir(). Try:

import board
print(dir(board)).

Kratos800, 2022-03-19:
code.py output:
['__class__', '__name__', 'A0', 'A1', 'A2', 'A3', 'GP0', 'GP1', 'GP10', 'GP11', 'GP12', 'GP13', 'GP14', 'GP15', 'GP16', 'GP17', 'GP18', 'GP19', 'GP2', 'GP20', 'GP21', 'GP22', 'GP23', 'GP24', 'GP25', 'GP26', 'GP26_A0', 'GP27', 'GP27_A1', 'GP28', 'GP28_A2', 'GP3', 'GP4', 'GP5', 'GP6', 'GP7', 'GP8', 'GP9', 'LED', 'SMPS_MODE', 'VBUS_SENSE', 'VOLTAGE_MONITOR', 'board_id']

robert-hh, 2022-03-19:
So no I2C in the list. You have toi get the I2C object from another class. As far as I recall, it's busio.
See https://docs.circuitpython.org/en/latest/shared-bindings/busio/
SCL and SDA are PIN objects of the pins, to which the device is connected. Please consult the documentation for further details.

Kratos800, 2022-03-19:
I tried with the busio but i am already using it for bmp280 sensor.

robert-hh, 2022-03-19:
I2C is a multi-device bus. You can connect multiple devices to it, as long as the device addresses are different.

Kratos800, 2022-03-19:
Thank you very much !! Its working now!

Kratos800, 2022-03-19:
I a again getting error:
ValueError: I2C peripheral in use

Kratos800, 2022-03-20:
This is the code in my main file it is in true loop 
uv_raw = UV.uv_raw
    print(uv_raw)

this is the code of the sensor (veml6070)

import code
import adafruit_veml6070

i2c = code.I2C()

uv = adafruit_veml6070.VEML6070(i2c)
uv_raw = uv.read

when i run the code in the main file it shows 
AttributeError: 'module' object has no attribute 'uv_raw'

robert-hh, 2022-03-20:
What is UV? How is it defined? Please make yourself familiar with the the scoping rules of Pyhton.

closed 2022-03-21 by Kratos800