robert-hh/SH1106 #21: Saving data

Kratos800, 2022-03-21:
I am using 2 pi pico board with circuitpython 
one is sending data and other is receiving data which I want to save in csv or txt file 

Can someone please tell me how to save data.  It is fine even if I am able to save data read by the board sending the data

robert-hh, 2022-03-21:
MicroPython supports the Python file operations to save data to a file. To convert data into a readable format, you can use json methods. Or you use the string formatting methods, like the % operator for strings of Python, or the format method for strings.

Kratos800, 2022-03-21:
I get this error:
OSError: [Errno 30] Read-only filesystem

robert-hh, 2022-03-21:
You have to disable file access from the PC to enable local file access. Otherwise you'll have file system corruption. Use the following commands to enable write access:

import storage
storage.remount("/", False)

See also the documentation: https://learn.adafruit.com/circuitpython-essentials/circuitpython-storage

closed 2022-04-19 by robert-hh