robert-hh/SH1106 #24: Constructor needs an id?

chard62, 2022-07-21:
I am using SH1106 library to use the OLED display with a Raspberry Pi Pico. The error I see is as follows:

MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
TypeError: 'id' argument required

In your documentation, I see no mention of "id". If I add "id=0" to the constructor, it works, I get no error.
My constructor is now: "i2c = I2C( id=0, sda=sda, scl=scl, freq=400000)"

robert-hh, 2022-07-22:
Some ports require an ID, some port do not. So it's better to take the I2C (and SPI) instantiation out of the SH1106 class and let the caller supply it. That follows the route of other classes for peripheral devices.

robert-hh, 2022-07-22:
The examples are for the ESP8266 port which does not required the id argument. Other ports like the RPi 2040 do. So that has to be adapted. Since the ESP8266 is rarely used now, maybe the examples have to be updated.

closed 2022-07-22 by robert-hh