robert-hh/SH1106 #25: Incorrect calculation of pages to update

peter-l5, 2022-08-25:
When using a framebuffer method with a h parameter the number of pages to update can sometimes be calculated incorrectly. This is especially an issue if clearing a row of text by writing a rectangle on top of it. 

So, for example, instead of:
    def rect(self, x, y, w, h, color):
        super().rect(x, y, w, h, color)
        self.register_updates(y, y+h)

the code should read, as follows (-1 added to third line):
    def rect(self, x, y, w, h, color):
        super().rect(x, y, w, h, color)
        self.register_updates(y, y+h-1)

CRImier, 2022-08-25:
you're right! want me to open a PR fixing the instances of that, or would you rather I PR it quickly?

peter-l5, 2022-08-25:
what's a PR?

CRImier, 2022-08-25:
pull request

peter-l5, 2022-08-25:
I could have a go

CRImier, 2022-08-25:
yay!
(fyi I'm not the maintainer of this driver, but I'm the one that made the page update calculation changes, which is why I'm replying)

peter-l5, 2022-08-25:
Perfect - I'll work on a fix.

peter-l5, 2022-08-30:
Hi,
To discuss this issue and propose a possible fix, I have created a branches in the fork [peter-l5/SH1106-dev-fork-issues-25-28](https://github.com/peter-l5/SH1106-dev-fork-issues-25-28) with the following purposes:
* to [demonstrate the issue](https://github.com/peter-l5/SH1106-dev-fork-issues-25-28/tree/fix25-demo) - this version includes an non-commented out line that prints which pages are updated - and shows that an additional page may un-necessarily be updated
* [a proposed fix](https://github.com/peter-l5/SH1106-dev-fork-issues-25-28/tree/fix25)
* to [demonstrate the fix](https://github.com/peter-l5/SH1106-dev-fork-issues-25-28/tree/fix25-demo) - this version also includes an non-commented out line that prints which pages are updated

The fix is the same as proposed initially, but that has not been built into the SH1106 master branch yet. (In doing this, I'm also practising/ learning how to use github better!)   

As discussed on the issue #27 thread, @robert-hh, I'll leave it to you to test these, but have produced some test code - as below. If OK then you could pull the fix, or I would be happy to create a pull request.

The proposed test code is as follows. I have run the testing part of the code on the [SH1107 driver](https://github.com/peter-l5/SH1107) that I have created from this one:
```
from machine import Pin, I2C
import sh1106
#
i2c = I2C(scl=Pin(5), sda=Pin(4), freq=400000)
display = sh1106.SH1106_I2C(128, 64, i2c, Pin(16), 0x3c)
display.sleep(False)
display.fill(0)
display.text('Testing 1', 0, 0, 1)
display.show()

# test code: these steps demonstrate that there is an issue when updating 8 pixels/ 1 page
time.sleep(1)
display.fill(0)
display.show()

print('vline')
display.vline(8, 0, 8, 1)
display.show()
time.sleep(1)
print('rect')
display.rect(16, 0, 24, 8, 1)
display.show()
time.sleep(1)
print('fill_rect')
display.fill_rect(16, 0, 24, 8, 1)
display.show()
time.sleep(5)

# these lines will demonstrate that when fixed 2 pages are still updated for objects 9 pixels high

print('vline')
display.vline(8, 0, 9, 1)
display.show()
time.sleep(1)
print('rect')
display.rect(16, 0, 24, 9, 1)
display.show()
time.sleep(1)
print('fill_rect')
display.fill_rect(16, 0, 24, 9, 1)
display.show()
time.sleep(5)

display.fill(0)
display.show()
display.poweroff()
```

peter-l5, 2022-09-01:
@robert-hh thanks for accepting/ pushing the proposed change. I think this issue is fixed, so is ready to closed.

closed 2022-09-02 by robert-hh