robert-hh/SH1106 #26: When the screen is rotated by 90 degrees the partial updates are not working properly

peter-l5, 2022-08-25:
Partial updates do not work properly when the screen is rotated by 90 degrees.
in normal orientation the display can be written in horizontal pages, 8 pixels high. 
if the framebuffer is effective rotated by 90 degrees by using colour mode: MONO_HMSB, then writing pages to the display writes vertical stripes of 8 pixels rather than horizontal ones. As a result partial updates for a rotated display do not work properly.
It is possible to fix this  by changing the display's memory addressing mode to vertical addressing and writing 1 row at a time.
I have written some code that implements this for the similar SH1107 display driver IC and will post it in the coming days.

CRImier, 2022-08-25:
ah nice! this isn't something I've seen people do, but having this be supported would indeed be lovely!

(for you or anyone who's reading this, you can disable partial updates by calling `.show(full_update=True)` if they're causing problems and you'd rather not dig deeper)

peter-l5, 2022-08-25:
I have placed some code that tackles this here: https://github.com/peter-l5/SH1107
There are quite a number of other changes in the code that speed it up slightly, at least according to my testing.
Unfortunately I think I have probably broken the SPI interface.
Any comments or suggestions welcome.
PS I'm pretty new to github so apologies if I'm breaking and conventions

peter-l5, 2022-08-25:
The changes include setting the I2C interface to run at 1MHz, which seems to work OK.
Ay this bus speed, full screen update takes about 33ms in normal orientation and about 58ms rotated by 90 degrees.

robert-hh, 2022-08-26:
The common style is NOT to change the interface properties in a driver that uses it. Interface setting should be left to the code calling it. In that case, the I2C object has to be created by the code using the SH1107 driver. Therefore it can be created as needed. The same applies to SPI.

peter-l5, 2022-08-26:
@robert-hh : thanks. My SH1107 driver is functional but still includes test code. The I2C frequency in the test code rather than the driver itself, which still allows the user of the driver to use an I2C object with a frequency property of their choice. I'd be happy to propose a fix via pull request for this issue based on the code I've written for the SH1107, although it might take 2-3 weeks.

closed 2022-09-02 by robert-hh