robert-hh/SH1106 #28: proposed enhancement to read a pixel's colour as well as set it

peter-l5, 2022-08-27:
Since at least micropython v1.9 the framebuf pixel function has had the ability to read a pixel colour as well as well as set it, see: https://docs.micropython.org/en/v1.9/pyboard/library/framebuf.html

Currently the implementation of the pixel function is as follows, which does not implement the ability to read a pixel.
    def pixel(self, x, y, color):
        super().pixel(x, y, color)
        page = y // 8
        self.pages_to_update |= 1 << page
        
To implement the read ability additionally, I would suggest the following formulation instead:
    def pixel(self, x, y, color=None):
        if color is None:
            return super().pixel(x, y)
        else:
            super().pixel(x, y , color)
            page = y // 8
            self.pages_to_update |= 1 << page

I have tested this possible enhancement but haven't implemented it as a pull request. Would this work?

peter-l5, 2022-08-27:
looks like the indentation has been lost, it should be as follows (now inserting as code):
`    def pixel(self, x, y, c=None):
        if c is None:
            return super().pixel(x, y)
        else:
            super().pixel(x, y , c)
            page = y // 8
            self.pages_to_update |= 1 << page`

peter-l5, 2022-08-27:
Hmm, that didn't work properly either: [I'm new-ish to github] apologies for additional messages!

robert-hh, 2022-08-27:
If you tested it you know whether it works. And if it is supported by framebuf, it should be supported by the sh1106 driver as well.

peter-l5, 2022-08-29:
@robert-hh The test that I have conducted is as of this [SH1107 driver](https://github.com/peter-l5/SH1107) which I have developed from this SH1106 one. I haven't directly been able to test the code on a SH1106 display as I do not currently have one, however as it only impacts the interface with the FrameBuffer module and not the SH1106 would you consider that an adequate test?

robert-hh, 2022-08-30:
If you post the test scripts, I can test them with a sh1106 here.

peter-l5, 2022-09-01:
@robert-hh thanks for accepting/ pushing the proposed change. I think this issue is fixed, so is ready to closed.

On the subject, I just thought I'd mention that it appears MicroPython is going to be upgraded with some changes to the [framebuf](https://docs.micropython.org/en/latest/library/framebuf.html) methods that will need some similar updates to that for `pixel()` in due course. (There are new `ellipse()` and ploygon `poly()` methods but no `fill_rect()` method any more.)

robert-hh, 2022-09-02:
The new methods are already available in the daily builds. And since all framebuf methods are available in subclasses like SH1106, you should be able to use them.

peter-l5, 2022-09-02:
Yes, good points. (I had assumed that many people might be used a distribution that is somewhat behind.) 
However wouldn't you lose the ability to draw filled rectangles? Because `fill_rect()` method would be broken and you would not be able to access the new ability of `rect()` to fill what it draws. 
Perhaps add an optional fill parameter to the `rect()` method?

robert-hh, 2022-09-02:
I do not know where you looked. But fill_rect() is still present:
```
>>> help(framebuf.FrameBuffer)
object <class 'FrameBuffer'> is of type type
  fill -- <function>
  fill_rect -- <function>
  pixel -- <function>
  hline -- <function>
  vline -- <function>
  rect -- <function>
  line -- <function>
  ellipse -- <function>
  poly -- <function>
  blit -- <function>
  scroll -- <function>
  text -- <function>
```

peter-l5, 2022-09-02:
I was looking here at the ["latest" version](https://docs.micropython.org/en/latest/library/framebuf.html) of the MicroPython documentation.

robert-hh, 2022-09-02:
Ok. So the plan is to het rid of fill_rect() sometimes and people should get used in using rect() with the optional 'fill' parameter.

peter-l5, 2022-09-02:
It looks like that, yes.

closed 2022-09-02 by robert-hh