robert-hh/SH1106 #33: Submit this driver to https://github.com/micropython/micropython-lib drivers

freemansoft, 2023-01-29:
Can you submit this file to https://github.com/micropython/micropython-lib/tree/master/micropython/drivers/display  Then it would be near the SSD1306 driver.

You could submit the file itself or ask if they would link it as a submodule like they do for many of the other mainline code like TinyUSB

robert-hh, 2023-01-29:
I could try that. There is a difference to TinyUSB. The latter is used for the USB support of the firmware and therefore mandatory. SSD1306.py or SH1106 are "just" an optional drivers, which can be used and copied to the board if needed.

freemansoft, 2023-01-29:
In theory, the easiest thing to do is to just submit the driver to micropython-lib as an addition and then maintain it there.   The micropython-lib team would probably just rather have everything reside there. 

You are right.  TinyUSB is different but really that particular library is cross-platform and used a bunch of places so I provided a lousy example.

I wanted to reference your driver file in a project without copying it in and without doing some fancy scripting.  I added it as a git module so your file gets pulled into my working directory when checking out my repo.  https://github.com/freemansoft/ESP8266-MicroPython/tree/main/freemometer