robert-hh/SH1106 #35: Adding driver to nano-gui PR #48

freemansoft, 2023-02-13:
I've submitted a slightly modified version of this driver to nano-gui in PR 48. The change basically lets nano-gui treat the device as a flattened color display so that the color code can run on monochrome displays.  nano-gui supported this. I just pasted in the appropriate boilerplate into the driver.

The PR has not yet been merged so there may be other changes.  This message as a courtesy so that you know this is happening.  It might make sense to add a link to nano-gui if the PR goes through, you are ok with it and think it is appropriate.

https://github.com/peterhinch/micropython-nano-gui
https://github.com/peterhinch/micropython-nano-gui/pull/48

robert-hh, 2023-02-15:
Thanks for mentioning. I'm fine with adding links to Peter's nano-gui.

freemansoft, 2023-02-18:
PR inbound

closed 2023-02-18 by freemansoft