robert-hh/SH1106 #42: multiple tracebacks with sh1106 oled screen

remiind, 2025-03-20:
Hi,

when using the driver, i run into this issue:

```
MPY: soft reboot
Warning: I2C(-1, ...) is deprecated, use SoftI2C(...) instead
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
  File "sh1106.py", line 173, in __init__
  File "sh1106.py", line 111, in __init__
  File "sh1106.py", line 116, in init_display
  File "sh1106.py", line 123, in poweron
  File "sh1106.py", line 178, in write_cmd
OSError: [Errno 19] ENODEV
```

This is the code in my main.py:

```
import time
import sh1106
from machine import I2C, Pin

i2c = I2C(scl=Pin(5), sda=Pin(4), freq=400000)
display = sh1106.SH1106_I2C(128, 64, i2c, Pin(16), 0x3c)
```

SCL and SDA are connected to 5 and 4.
sh1106.py driver is in the root directory of the esp32.

robert-hh, 2025-03-20:
The significant error is:
```
File "sh1106.py", line 178, in write_cmd
OSError: [Errno 19] ENODEV
```
It tells that the I2C device is not detected. Possible reasons:
- Missing pull-up resistors
- wrong device type or address

What is the result to:
```
from machine import I2C, Pin
i2c = I2C(0, scl=Pin(5), sda=Pin(4), freq=400000)
print(i2c.scan())
```
The SH1106 can be configured for different I2C addresses, and some of them support either SPI or I2C.

remiind, 2025-03-20:
Result:

```
MPY: soft reboot
[]
```

remiind, 2025-03-21:
Hi Robert,

the new screen works like a charm.

Thank you!

robert-hh, 2025-03-22:
That's good. So I can close this topic now.

closed 2025-03-22 by robert-hh